Sienci-Labs/gsender #452: gsender 1.3.10 EDGE misidentifies certain COM ports as grblHAL

kbulgrien, 2024-01-20:
I have an [HP MP9 G2 Retail System](https://support.hp.com/us-en/document/c04899268) loaded with Windows 10 Pro 22H2.

* An Inventables X-Carve 1000mm x 1000mm CNC router w/ X-Controller is connected to one of the USB ports and is shown as **USB Serial Port (COM4)**.
* On this system, there is also an **Intel(R) Active Mangement Technology - SOL (COM3)** device shown under **Ports (COM & LPT)** in the **Device Manager**.

[image]

If one starts gSender 1.3.10 EDGE in this configuration, upon startup, when one pulls down "Connect to Machine", there are listed:

+ Recognized Devices
   - COM4 / use grblHAL at 15200 baud
+ Network Devices
   - IP-Address / Ethernet at IP-Address:23
+ Unrecognized devices (1)
   - (->) / Grbl | grblHAL

[image]

Clicking **(->)** reveals additional details:

+ Unrecognized devices (1)
   - COM3 / use grblHAL at 15200 baud

[image]

If one "connects" to COM3, the UI says "Connected COM3/grblHAL" at the top of the screen, but in the visualizer, is shows "Disconnected".

[image]

This is seemingly somewhat undesirable.  I can provide additional information about this system or run additional tests as directed.

kglovern, 2024-01-22:
This is working as expected.

There is no way to determine which com ports are actually controllers without connecting to them in turn and checking.

Our solution, and what you're seeing, is filtering based on recognized device product and vendor IDs since the # of subsets of available Arduinos/STM microcontrollers is small.  So devices we know could be grbl devices appear as recognized, and anything else (such as your internal device) is show as unrecognized.  

Unrecognized devices has to exist because theoretically people could be connecting to a GRBL/hal-adjacent machine that we don't currently have in the filter - and if they didn't have a way to see it if unrecognized they'd never actually be able to connect to it.

The connection widget shows connected as to a serial port (because it is, serial connection isn't dependent on what the device is) while the machine status shows disconnect (because there is not a valid firmware on whatever you're connected to).

You'll see in a number of senders just a list of all available com ports and it's up to the user to connect to the correct device.  It's the same thing here except we've taking some steps to highlight the devices that we know could be GRBL/hal devices which usually makes the choice obvious (not too many people have multiple arduinos plugged in)

kbulgrien, 2024-01-22:
Ok, that sounds reasonable, except that I still question the **grblHAL** indication.  To me, it makes it sound as if it WAS connected to and WAS identified as grblHAL.  That is how this issue came up.  I connected to it to see if it was a mirror of my COM4, because it made me think a second device was actually created for the controller.

kglovern, 2024-01-22:
grbl/grblHAL indicates which controller type to connect using - hence the "Use" part.  It's just an indication of which controller type to connect to which port.  You'll notice is swaps based on the selected firmware flavour toggle consistently across both types.

closed 2024-01-23 by kglovern

SienciLabs, 2024-11-05:
Yeah @kbulgrien I can understand what you're saying. I think ultimately it's more an issue of UI communication that hopefully on our gSender redesign we can make clearer. We mainly want to communicate the currently selected baud and firmware, then display the list of COM ports, but the design right now makes it so all available ports look like potentially valid controllers because of the label next to them

Thanks for the feedback, we'll keep our eyes out for it moving forward 🙏