Sienci-Labs/gsender #456: 1.4 quick corner buttons slams into limits

hamanjam, 2024-01-25:
Installed 1.4 when it was released and noticed off using the quick corner buttons, the commands are different from previous versions.  I edited my maxium Y travel (previously 861) thinking my sensor moved but also noticed it does the same thing on X.    I have inductive sensors and dust shields and previously adjusted maximum travel to give the greatest work area.    I see right off that the corner buttons are enough to be an issue.  I can hear the motors skip when they hit the limits with 1.4:

1.3.10 back left:  G53 G21 G0 X1 Y807.500
1.4 back left: G53 G21 G0 X5 Y845.000

1.3.10 back right:  G53 G21 G0 X775.200 Y807.500
1.4 back right: G53 G21 G0 X811.000 Y845.000

------------------
$100=201.715 (X-axis travel resolution, step/mm)
$101=200.523 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$110=4000.000 (X-axis maximum rate, mm/min)
$111=4000.000 (Y-axis maximum rate, mm/min)
$112=3000.000 (Z-axis maximum rate, mm/min)
$120=750.000 (X-axis acceleration, mm/sec^2)
$121=750.000 (Y-axis acceleration, mm/sec^2)
$122=500.000 (Z-axis acceleration, mm/sec^2)
$130=816.000 (X-axis maximum travel, mm)
$131=850.000 (Y-axis maximum travel, mm)
$132=121.000 (Z-axis maximum travel, mm)

hamanjam, 2024-01-25:
I recorded videos of each version if it helps.  I can switch back and forth between Edge 1.3.10 and 1.4

kglovern, 2024-01-25:
One of the changes made for this release was making corner buttons more consistent - it was formerly EEPROM pulloff distance for homing corners and 95% of machine limit on extents.  This was changed to a consistent value all around.  

How far is it torquing out?  We picked a consistent value away from the machine limits, but might end up changing it back to a % of travel in all limits as a bit of a safety buffer.

SienciLabs, 2024-01-26:
Hey @hamanjam, what I'm confident is happening here is that the corner buttons on past versions were much more generous with the distance they were spaced from the machine limits, as Kevin mentioned. This means that if you set up your machine so the corner buttons went all the way to the corner, then on 1.4.0 it will now go further than then because we reduced the offsets from each limit.

This has also revealed to me though that the defaults we have for the LongMill profiles are likely over-ambitious since once a limit switch is installed the amount of machine travel can be reduced by 10 or more millimeters, this would cause a LongMill with default settings to travel past the machine travel when using the corner buttons. We'll be updating the default travel distances for all machines moving forward to compensate for this 'inaccurate' profile we've been shipping all machines with to-date

hamanjam, 2024-01-26:
I'm going to backpedal here and close this as an issue.  1.4.0 has simply brought out that I had the max travel set incorrectly.  I'm assuming that a nudge of the Y limit and adding dust shields changed my work area slightly (enough to be called out by the new release).  I would agree that a "buffer" may be necessary because ID10T like me are out there.   

I manually jogged XY to the physical limits and found my free working area is actually X807 Y857 and when I set that in firmware, I can jog to both back corners without the "crash"

closed 2024-01-26 by hamanjam

kglovern, 2024-01-26:
FWIW, we've updated our default values for machine travel and will be rolling those out both in gSender and future firmware builds so thanks for bringing it up.