Sienci-Labs/gsender #464: Remote session has no access to settings

rcp1, 2024-01-28:
Hi,

a click on the settings button in a remote sesssion will just open a black screen which one can not exit anymore. Tested on Windows 11 and Linux Debian Bookworm.

All other menus like Surfacing or Firmware work fine...

rcp1, 2024-01-29:
Update: Works fine if remote client uses Edge, above problem only occurs in Firefox 122.0.

JohnOCFII, 2024-01-30:
> Update: Works fine if remote client uses Edge, above problem only occurs in Firefox 122.0.

Interesting.  I can confirm this issue on both gSender 1.3.10 Edge and 1.4.1 with FireFox 122.0 on macOS.  The issue does not occur using Safari 17.3 on macOS Sonoma 14.3.

kglovern, 2024-01-30:
The problem is firefox specifically with gamepad library - we'll look into it.

kglovern, 2024-02-05:
This is fixed and will be part of the next release.

closed 2024-02-05 by kglovern