Sienci-Labs/gsender #467: 1.4.1 throwing modal errors possibly surrounding spindle delay

hamanjam, 2024-01-31:
Getting an intermittent issue where I start a job and it pauses on a modal error.  If I restart gSender and run the same file, it works with no issue (but that brings up another possible intermittent problem losing my program start/stop commands)  My gcode file includes the following.  The error will randomly either present on the spindle speed line or the next line after

---------------
T1
G17
G21
G90
G0Z6.350
G0X0.000Y0.000
S22000M3 G4P7
G0X202.435Y45.983Z2.540

----------------

[image]

hamanjam, 2024-01-31:
On a different file, the error throws the line after the spindle speed line:

-------------------------
T1
G17
G21
G90
G0Z6.350
G0X0.000Y0.000
S27000M3 G4P7
G0X196.876Y41.798Z2.540
-------------------------
[image]

kglovern, 2024-01-31:
We'll have to make sure we're not adding G4 if spindle delay is enabled in preferences if the line already contains a G4 - you can see the double command, G4P7 from your post-processor and G4P1 from us.

We'll get it fixed but for now disabling spindle delay in settings if it's already being added by post-processor should fix the error issue.

hamanjam, 2024-01-31:
I saw that it's doubling but only intermittently.  The regex I submitted as part of issue 439 should account for adding the G4 line only if it doesn't already exist which looks like the normal operation.  I'll try to run some more scenarios to see if I can find a pattern as to when it does/doesn't add the additional G4

kglovern, 2024-02-02:
Should have a fix for this next release.

Debug flags are dev only unfortunately.

hamanjam, 2024-02-24:
Appears to be resolved in 1.4.3