Sienci-Labs/gsender #471: 1.4.1 doesn't respond to jog controls in main UI

cvandermey, 2024-02-10:
in 1.2.x my logitech controller works fine to jog the machine.  This is OSX 12.7.1.  In gsender 1.4 and 1.4.1 the settings recognizes the controller and displays the commands from it, but in the main UI the controller shows as "Unknown" instead of "Logitech..."  and fails to jog the machine.

walidkayhan, 2024-02-12:
Not sure what you mean by:

> but in the main UI the controller shows as "Unknown" instead of "Logitech..." and fails to jog the machine

Are you referring to the popup notification that shows up on the bottom left of the application when you first connect your controller?

cvandermey, 2024-02-12:
yes, exactly.  This to me seems to indicate that there is some problem
registering the device.

walidkayhan, 2024-02-12:
I see, the gamepad profile for your controller might be corrupted in the settings. We made some big changes to gamepad support between v1.2.x and v1.4.x, so some people may experience some migration issues in certain areas.

To fix this problem, try deleting that gamepad profile and setting up a new one with your controller.

Let me know if that fixes it for you.

walidkayhan, 2024-02-21:
Will be closing this for now, please reach out if you are still experiencing this particular issue 😃

closed 2024-02-21 by walidkayhan