Sienci-Labs/gsender #473: Sending $$ to fluidnc crashes gsender

AlexSBrown8192, 2024-02-20:
Firstly well done on an excellent program. I have everything working superbly with my home built fluidNC controller. The only issue I have is when an alarm is triggered and the red flashing box appears. Clicking this sends the correct reset signal but this is followed by a $$ command. As fluidNC does not send back the full grbl settings list ( only $10 and $30) gsender crashes to black screen requiring restart. Is there a way to prevent this being sent? I am using v1.4.1 and the generic machine profile. Thx in advance

AlexSBrown8192, 2024-02-29:
After looking through the code my issue may be resolved by removing line 1071 in gsender/src/server/controllers/grbl but I have no idea how to compile this as I have no experience in this field. I am a MCU and plc programmer with no js knowledge. Apologies for my ignorance

hamanjam, 2024-02-29:
My concern would be more of why the controller isn’t grbl compliant and returning all firmware values.  Without reading back your settings from the controller, gSender doesn’t know the limits of your machine or and accessories such as touch plate pins, homing sensors or even motor steps per inch/mm.  I would think without that info returning from the controller, I could see more issues trying to get it to work properly.

Message ID: ***@***.***>

AlexSBrown8192, 2024-02-29:
Thank you hamanjam. It's certainly strange but everything works perfectly well apart from sending $$. I think it's because fluid nc handles all the control routines and reports status etc back to the sending program. I can control the machine directly via WiFi without any sending software but the web server is still a rather clunky interface compared to gsender which is far superior to use. Anyway it's a rather minor issue and I have a workaround should an error occur.

AlexSBrown8192, 2024-02-29:
Installed 1.4.3 today. Still freezing on $$ but no black screen.

AlexSBrown8192, 2024-02-29:
Understood. Many thanks I'll look into that. Thank you for your time

hamanjam, 2024-02-29:
I've searched a bit and it seems common with a lot of gcode sender programs.  Openbuilds flat out says they have no plans to support fluidNC since their boards are grbl.  I see it difficult for the dev team to support a board they don't offer especially when they are already maintaining grbl and grblHAL. 

It's open source and maybe someone is working on a fork that does support your board but $$ is a fundamental grbl command and if the program doesn't get a full result, it will simply wait until the expected response. I could envision maybe a "timeout" routine which would just throw an error you can acknowledge and move on if the response doesn't return in a set time but removing the grbl info request would potentially break the rest of the world that uses grbl 1.1 boards.

I'm not disputing the value of the request and I see that the comparable apps such as bcnc may not be as polished.  I just don't know how realistic it would be to take on another hardware stream

AlexSBrown8192, 2024-03-02:
Thank you hamanjam I agree completely. I am having great success so far with only minor issues. I will create a list of problems I encounter and any workarounds I find as time goes on. Hopefully this will assist any others that try this combination. Thanks to all for your input

closed 2024-07-15 by kglovern