Sienci-Labs/gsender #474: New version thinks my machine is locked all the time

Windcrest77, 2024-02-23:
I made the mistake of upgrading and now I'm dead in the water. When I connect my machine, it connects but I dont get the button that says run homing. Instead it says the machine is locked, its not locked, and the unlock button does nothing. How can I revert back to the previous version of gsender.

Windcrest77, 2024-02-23:
An update... I happened to have the installer for v 1,2,0 still in my downloads folder so I installed it. This version is much better than the latest. When I connect to the machine it gives be the Home Machine button. It doesnt think the machine is locked like the current version. It should work for me even with its old bugs. This proves that the latest version has a serious issue upon startup thinking the machine is locked. This is a showstopper bug that got introduced here.

hamanjam, 2024-02-24:
Have you tried just jogging the machine?   The “lock” icon is always there.

Make sure when you connect, you need to make sure you connect as grbl, *NOT* grblHAL.

hamanjam, 2024-02-24:
To revert to the older versions, follow this thread

https://forum.sienci.com/t/how-do-i-revert-versions-in-gsender/9100

Windcrest77, 2024-02-24:
> Have you tried just jogging the machine? The “lock” icon is always there. Make sure when you connect, you need to make sure you connect as grbl, *NOT* grblHAL.

Yes I've tried everything, I can't jog via software if it thinks its locked. I manually cranked each leadscrew, clicked all the limit switches. Same result. But if I uninstall the latest version and install 1.2.0 it works great. Its definitely a software issue. Of course I picked grbl, its on COM5 with my machine, old version works great, new version is bricked right out of the gate. Also when I do connect, it connects much slower than the old version. 5 seconds with the old ver and I get the homing alert and go. With the new version it takes 20 maybe 30 seconds then I get the locked alert and it won't unlock, all axis are centered well away from any limit switches.

Windcrest77, 2024-02-24:
Also the version I installed was the version offered from the Scienci website.

hamanjam, 2024-02-24:
About 75% of the problems people have been having is due to selecting grblHAL so it was a legitimate question.   I'm using 1.4.3 with no movement issues (same with previous 1.4.x versions) so it is not a "showstopper" software issue.  If the software can't read your firmware settings, it cannot determine the status of your machine so it won't move or present the home button.  It has nothing to do with the lock/unlock button

Windcrest77, 2024-02-24:
I have a very popular type of machine a ubiquitous 3020 Chinese machine. GSender never had a problem reading it until now. I would imagine these 3020's are out there more than any machine. Also the Candle sender that the machine came with works, as well as UGS, but I really like the UI of GSender. So we have previous versions of GSender that work fine and other sendedr programs that work fine. But the most recent GSender is broken, I would call that a showstopper for GSender considering that these 3020's are about as generic as you can get in a CNC.

This model:

https://www.amazon.com/gp/product/B0BL3YYQVN/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Windcrest77, 2024-02-24:
More info that might be helpful. I installed version 1.3.10 and it works fine. This was the last release prior to the 1.4 versions. So I would safely say that this bug was introduced in one of the four iterations of 1.4 so far. Thanks for the help and attention here, I dont mean to be a pest, just reporting this very show-stoppy bug with the most basic machine on the market. If it will help the developers I would be happy to re-install the latest and make a video demonstrating the bug. Meanwhile I can go back to making chips using the last 1.3.10 version which connects fast and doesnt put up the locked warning. If you want me to test the next release if you think you have a fix, let me know I can find time for that, as I want to help with this elegant program.

Windcrest77, 2024-02-24:
Another update. Version 1.3.10 fixed the original locking bug. But when I went further I found that the Z probe did not work at all. It put up the message "Intiating probing", but the machine stood still. So I installed version 1.4.0, this version does not have the serious unlock bug like the latest ver, and the probe works again. I'm going to do some cutting now, lets hope for the best. I'll report if I have problems with 1.4.0.

Windcrest77, 2024-02-24:
Another update.  I reverted back to old reliable version 1.2.2. Version 1.4.0 had a bad bug. It actually runs the job in a different place on the work surface than the Outline said it would. I rehomed several times to make sure I wasnt imagining it. When I did an Outline I see that the bit will travel exactly where expected. But when I actually run the job it starts cutting about 200 mm shifted on the X axis. To the point where it would be hitting my clamps. So I went back to 1.2.2, same job, same procedure, everything the same, 1.2.2 cuts the job exactly where the Outline said it would. I will be remaining at 1.2.2 until such time that I have the time to try other versions again.

hamanjam, 2024-02-24:
I can almost guarantee that you connected as grblHAL because I can reproduce that behavior exactly when I connect a grbl board with grblHAL.   The only way to be sure would be to provide screenshots of the connection tab in 1.4.3 while connected as well as the firmware screen.

Windcrest77, 2024-02-25:
At some point i will try again. On my machine the grblHAL had what looked like an IP address not the COM5 I'm used to seeing, so I never tried that one not knowing what it was. This is my basement machine a 3020 $600 Chinese model 300mm x 300mm on which I use a 1.25 HP Makita router as the spindle. I machine custom aluminum heatsinks for a bulk heatsink supplier when his customers want machining of holes, routing down fins, etc. When I get caught up I will try again as I do want the latest updates.

hamanjam, 2024-02-25:
GrblHAL will connect either COM or Ethernet.   When grbl is selected, only COM will show.  If you saw Ethernet as an option, you had selected hrblHAL as I said

[image1.jpeg]

[image0.jpeg]

hamanjam, 2024-02-25:
grbl
[image]

grblHAL
[image]

closed 2024-03-28 by kglovern