Sienci-Labs/gsender #476: discardVirtualMemory not found since unisntall old version to version 1.4.3

JEOBDX, 2024-03-02:
Hi everyone,

Since I uninstall version 1.2.2 ( it worked perfectly) to update with 1.4.3 version ,I have an error 
"the DiscardVirtualMemory procedure entry point was not found in the dynamic link library C:\Program files\gSender\gSender.exe "

Maybe issue come from my windows version 8.1 ?
 Do you have any idea ? 
Thank you

kglovern, 2024-03-02:
We no longer support Windows 7/8 as they have reach EoL support from Microsoft.  It was a tough decision but the percentage of users on Windows 7/8 was sub 1% of total users and not updating our libraries that dropped support (Electron/Serialport) opened us up to security vulnerabilities.

closed 2024-03-02 by kglovern

JEOBDX, 2024-03-02:
Ah OK !!  I understand now.....
So I part of less 1% .....   with my old laptop witch I use only for CNC .

Thank you for your answer , just update it now !