Sienci-Labs/gsender #477: Zeroing causes probe to plunge beyond zero point

cvandermey, 2024-03-03:
1.4.3.  Autozero touchplate, 1/4 bit using "auto".  Running "probe" touches off correctly, then moves to x/y 0, then slowly plunges into the spoilboard continuously (gsender reports -0.97" depth before it throws an error.)

bigbearct, 2024-03-06:
Had the same experience , nearly broke the bit .

kglovern, 2024-03-13:
We were able to reproduce a separate issue with auto-zero probing related to $13 enabled, but weren't able to recreate this issue specifically.

Do you have $13 enabled in your EEPROM settings?  If you disable it, do you see the same behaviour?

kglovern, 2024-03-15:
1.4.4 has fixes to autoZero with $13 enabled related to X and Y movement - let us know if you're still having issues with the Z.  Again, we were unable to reproduce that specifically while identifying other issues.

kglovern, 2024-04-01:
It's been 2 weeks without further comments - if this is still an issue, re open another issue.