Sienci-Labs/gsender #478: 1.4.3 Rapid/Precise jogging presets don't work

cvandermey, 2024-03-03:
Normal preset in gsender works for jogging operations - z/x/y.  However, when selecting rapid/precise, the machine doesn't respond.  I see error 22 in the console: feeder $J=G20G91 Y0 F0
error:22 (Undefined feed rate)

bigbearct, 2024-03-06:
Also doesn't seem to save them from job to job. Precise would go either really really precise and rapid would also go very slow until I would default the settings and then seemed to clear it up until I setup a new job and have to go thru it again.

kglovern, 2024-03-14:
This should be resolved with next release.

kglovern, 2024-03-15:
Let us know if 1.4.4 has resolved these issues for you.

bigbearct, 2024-03-20:
1.4.4 still doesn't save or hold the settings. I usually have to do a restore settings for the 3 different speeds to work correctly.

walidkayhan, 2024-04-12:
@bigbearct are you still facing this issue on the latest version, v1.4.6?

bigbearct, 2024-04-12:
no issues since going 1.4.6

walidkayhan, 2024-04-12:
That's great to hear 😃  if you notice another problem with jogging presets or anything else please open another issue

closed 2024-04-12 by walidkayhan