Sienci-Labs/gsender #482: Native build for Mac M1/ARM?

Tempest2k8, 2024-03-08:
I would like to see a newer build supporting the M1/ARM architecture or perhaps some documentation on how we can build it from source ourselves. There were previous builds for the M1 but now it looks like only Intel is supported.

walidkayhan, 2024-03-12:
We had plans to add the macOS ARM64 build for gSender a little while back but we unfortunately need to push it back for now as we have other features to focus on. We will add it to the patch notes once ARM64 is part of our build process :)

J-eremy, 2024-03-20:
> I would like to see a newer build supporting the M1/ARM architecture or perhaps some documentation on how we can build it from source ourselves. There were previous builds for the M1 but now it looks like only Intel is supported.

The entire program is a node/react app wrapped in electron. You should be able to build a functional copy yourself from source without much fuss, but some fuss none the less.

It gets rather complicated as it is a client/server architecture where the server talks to the machine and the react client talks to the server, then both are wrapped in electron.

If you don't want it wrapped in electron then its even easier, as it can be run with just node.js and leave the chromium bloat completely behind.

Also, what I have witnessed that without electron the whole package only uses about 100mb of ram, compared the the 16 megatons of ram chromium demands.

The current build process is kind of messy right now for a few reasons but can be done. Documentation or a wiki page would be much appreciated for the process. I keep all my notes in an obsidian notebook documenting all my progress with building this thing but have no clue on where I would even submit them for others.

A few hints tho from my experience,

- the package manager used is yarn.
- it has to be built with node 14.17.6
- the dependency for `"electron": "^26.2.2",` is listed as  dev dependency in the main package.json, but is required in the src package.json because of the sentry module used for data collection. which halts the install of dependencies to build it standalone. 
- After its built, It has to be run with node 21.6.2 or it throws more errors.

There are a lot of idiosyncrasies because the developers clearly aren't focused on people building it themselves, which is ok, as the mass base of the users will never need to.

Saying all that, the reason why there probably isnt any documentation on building it is exactly those reasons above. And steps like that would confuse more than help.

I think your best bet would be to run it all in a docker container, Im working on building a dockerfile to do all the things needed to build it but time is limited.

closed 2024-07-15 by kglovern