Sienci-Labs/gsender #485: Gsender doesnt play well with lower resolutions.

J-eremy, 2024-03-12:
With version 1.4.3 a problem has appeared in the UI that did not exist in previous versions. At least not on v1.4.1

The laptop I use has a low resolution of 1366x768 and the adjusters for the move increments and speed are not usable anymore.

[image]

J-eremy, 2024-03-12:
This issue seems to be resolved as of 1.4.4-RC1. I will leave this open for anyone having this issue until 1.4.4 goes release.
Thanks everyone!

kglovern, 2024-03-13:
Yes, as mentioned we've already fixed it and it should be part of the next release 👍

kglovern, 2024-03-15:
This is now released as of 1.4.4 👍

closed 2024-03-15 by kglovern