Sienci-Labs/gsender #486: Remote connection doesnt have quick jog location unless re-homed

J-eremy, 2024-03-12:
If you connect to and home the machine from the connected device the quick jog buttons for home, top-left, bottom-left, and bottom right work as expected.

If you connect remotely, everything seems to function except the buttons for those quick locations are grayed out, until you home the machine from the browser on the remote device.

windows laptop directly connected to the machine view:
[image]

remote device after connection
[image]

I use the remote feature a lot, as gsender is running on a laptop with low resolution (1366x768)
not a huge deal as I can just home it after I connect just figured I would let everyone know.

kglovern, 2024-03-13:
We can take a look for probably the release after this one 👍

walidkayhan, 2025-07-07:
This has been fixed as of gSender v1.5.1. If you notice this bug has persisted for you, feel free to open a new issue :)

closed 2025-07-07 by walidkayhan