Sienci-Labs/gsender #487: Request: execute gcode with A axis

vazhure, 2024-03-13:
Please add the feature to execute gcode with A axis. Visualisation not matter for me.
I have 4 axis GBBL board and I can't use gSender with "A" commands and macros... 😞

vazhure, 2024-03-18:
It's a sad thing to hear, but it's not a difficult request. Just broadcast A commands from file to the grbl. 😞
Think I should start working on creating my own application...

walidkayhan, 2024-03-18:
I will keep it on my list of nice-to-have features :) but I don't want to give the expectation that it'll be in the next few versions of gsender soon because it likely won't be, unfortunately. You are more than welcome to open a pull request with that particular feature you want and we'd be more than happy to merge it in!

SienciLabs, 2025-07-17:
@vazhure I'll merge this into #584 as a singular feature request for A-axis control in grbl 👍

closed 2025-07-17 by SienciLabs