Sienci-Labs/gsender #491: Heightmap

mohammad787, 2024-03-20:
Would you please add height mapping to the app

J-eremy, 2024-03-20:
I second this, I know it was mentioned a while ago and there was even a menu for it at one time, and I also understand how much work it would be but this sentiment is also on my list. lol

patofoto, 2024-04-11:
Yes!!!. Gender would be perfect if we could build surface meshes. I need this for PCB milling.

apatieh, 2024-04-11:
I used to use candle but I'm using gsender for a while as it improved so much but it lacks this feature.
please add it

SienciLabs, 2025-03-01:
Closed as duplicate of https://github.com/Sienci-Labs/gsender/issues/74

Technically this one came sooner, but the other has since had more conversation in it

closed 2025-03-01 by SienciLabs