Sienci-Labs/gsender #493: feature - disable job summary

gcormier, 2024-03-23:
When I'm batching a ton of coaster jobs that take about 45 seconds each, the job summary is one extra popup I don't need at the end of the program.

Would be great to have an option to disable job summary. Bonus points, maybe disable job summary when job is less then X minutes in duration (also configurable).

hamanjam, 2024-03-25:
enable/disable the summary would be nice.  We can always see the job history if we need to look back at a job

BaysideCustomTC, 2024-03-25:
+1 vote for the option to disable. The Run Length is visible at the bottom of the screen, and the new Job History & Stats shows everything that we might need.

kglovern, 2024-03-25:
We'll see what we can do.

J-eremy, 2024-05-02:
Currently the job summary isn't hard bound to any confirmation. You can click anywhere on the window to make it disappear. Essentially you can just pretend it doesn't exist right now.

kglovern, 2024-05-09:
It's pretty painless to clear from the screen yeah, but we agree that some people might not be interested in ever seeing it.  We're adding a preference to disable this (on by default) in the next release.

closed 2024-05-09 by kglovern