Sienci-Labs/gsender #494: Workspace changing to P1 after job completion

BaysideCustomTC, 2024-03-25:
I sometimes run jobs in separate workspaces, each with their own XYZ0 origins.

After running a job in a workspace other than G54 (P1) e.g. G58 (P5), the workspace is set back to P1. If not noticed / corrected, then unwanted results can occur. For example, after completing a job I often move the spindle back to XY0 / Z39 as I know that to be a safe space. Obviously in P1's workspace that may not be a safe space for P5, causing collisions with clamps or other pieces of material.

I would expect that the workspace remains as P5, rather than being changed back to P1.

Please note that this is not a 1.4.4 specific behaviour - it operates the same in v1.1.7 and earlier.

[image]
[image]

hamanjam, 2024-03-25:
This is expected behavior.  The end of your file most likely has an M2 command.  See this link:

https://resources.sienci.com/view/gs-feedback/#workspace-is-changing-when-a-job-is-run

BaysideCustomTC, 2024-03-25:
It does indeed have an M2 command, as I would have expected. I amended the Program Start & Program Stop Events per the article linked and the behaviour is now as I would like. Thanks @hamanjam

closed 2024-03-25 by kglovern