Sienci-Labs/gsender #495: Option in Go To function to not go to Z

BaysideCustomTC, 2024-03-25:
The Go To function is quite useful.

However, when I tried it out, I saw but foolishly ignored that Z 0 setting.

I would like to see a checkbox / radio button alongside the Z [with default unchecked] which controls whether the G90/G91 command includes the Z instruction or not.

kglovern, 2024-04-01:
Would it be more useful to have an option to toggle off specific axes or for absolute to have it pre-populated with current workspace position (so you can change what you want and not have to worry about non-changed axes moving) and relative prepopulated with 0's?

BaysideCustomTC, 2024-04-02:
> Would it be more useful to have an option to toggle off specific axes or for absolute to have it pre-populated with current workspace position (so you can change what you want and not have to worry about non-changed axes moving) and relative prepopulated with 0's?

Yes, that would be a good idea. Default option would be Absolute / X & Y enabled / Z disabled / populated with current X/Y/Z positions. If it could remember previous settings (e.g. I last moved just X then just have X enabled next time I use this feature). Defaults X/Y/Z for Relative = 0/0/0.

kglovern, 2024-04-30:
This behaviour has been changed as of 1.4.7.

closed 2024-04-30 by kglovern