Sienci-Labs/gsender #497: feature - positioning with a camera

lucem, 2024-03-25:
Hi, I also posted this as a comment on this video (https://www.youtube.com/watch?v=-PpH7bcs7vM),
but since you asked me to, I'll open it up as a feature request on github as well.

bCNC has a nifty camera alignment plugin (https://github.com/vlachoudis/bCNC/wiki/Probe-Camera-Alignment)
which would suit gSender well if it was implemented. (Here it is in motion https://www.youtube.com/watch?v=bs4og_uanjM)

Essentially, it consists of grabbing a frame of the camera (regular webcam), overlaying the crosshair graphic with correct scaling
(explained in the bCNC wiki entry) and knowing the fixed tool to camera offset in XY; that way, you can move using the camera and
align to the stock visually, set your zero, then switch over to the tool and the center of the tool is in the exact place where the camera has been before.

In terms of effort, it's not that hard to implement. I'd do it myself, were it not for the fact that
a) I am no web developer, I write C++
b) I have no experience programming Electron, which is essentially a browser (chromium) and a lot of APIs on top of it
so I think it would take me way too long to come up with an implementation and send you a pull request.
Instead, I will ask you to consider implementing it.

For me personally, the lack of being able to use the camera is what keeps me from using gSender exclusively as an otherwise excellent
gcode sender, as in other ways, it is more than perfect - grblHAL support as a convincing argument for instance :)

Thanks a lot, I hope you find this feature request as useful as I do.

stefan63, 2024-08-23:
Yes we need this feature for PCB drilling, please!

J-eremy, 2024-08-29:
Im gonna bump this strictly for the reason that this would also help with fixture/clamp locating for some of the saucy clamping jobs.

lucem, 2024-10-21:
Since it has been a long time already and there has been no reaction from the developers, I am going to bump this one time.
It still is an important enhancement to make.

SienciLabs, 2024-11-05:
Thanks everyone, we've been bearing down on the gSender remake since it was a priority to address libraries that were losing support and the integrity of our build process. This sort of stuff is ready to be back on our list once we're done finishing our current grind 🙏

stefan63, 2024-11-05:
Love to see this happening, Gsender is by far the most apealing and also best looking UI for grbl that I have come across so far. 
Keep up the good work and thank you for all the work already put in to this software!