Sienci-Labs/gsender #498: Diagonal jogging not working intuitively

TITAN3737, 2024-04-08:
Diagonal controls in the UI works differently when clicked once and when holded continuously.
When any of the button diagonal blue arrows is clikced once, it moves the machine exactly one in each direction of the XY move distance.
But if clicked and hold (continuously), it does not moves diagonaly but it goes toward the corner of the working area. 
While I understand this maybe in some cases useful, it is pretty confusing that these buttons behave so differently. In the Wiki page is nothing about this feature.
Would it be possible to either make this feature optional or implement some kind of toggle for who want to use it only when needed?

TITAN3737, 2024-04-10:
Thank you for your reply.
> I'm going to assume you have soft limits enabled because that's the only way this makes sense.

I do, makes sense to me as useful safety feature.

From what I can see in the console, there is hardcoded 10000mm distance in each axis when using continuous jogging. So that makes every move to go out of the work area no matter where the machine acutely is. Hence this odd behavior of pointing the machine in different directions based on its current location.
Can these 10000mm be calculated to be the smallest distance to edge of work area? Those are work area lengths minus current machine coordinates in that given diagonal direction. For example: Work area x1000y1000, machine coordinates x600y500 / in case of diagonal move to top right corner, than smaller of x1000-600, y1000-500 / than $J=X400Y400F5000 ?
Or is it something more to this issue?

kglovern, 2024-04-10:
That will not work without homing switches and we can't assume every machine has homing switches.  Without homing switches, machine 0 is wherever the gantry is located when connected - which means we can't rely on EEPROM travel.

TITAN3737, 2024-04-10:
I was always wondering what are machine without homing good for, but from what I can tell, without homing and knowing where the tool is in the machine space, there has be very little accuracy overall. So I guess who do not need this precision, does not care about being diagonal move really in diagonal direction anyway.
So, can this be the toggle then? I mean no homing = corner arrow sends the machine roughly in that direction, and with homing the solution I have proposed.
Only beacuse some machines dont have homing, cannot mean that everybody has to use this inconsistent feature.

closed 2024-04-12 by kglovern