Sienci-Labs/gsender #500: SLB disconnects when homing.

hamanjam, 2024-04-09:
Using both USB and Ethernet, when I home the SLB using gSender 1.4.6,  The machine disconnect upon the X hitting home.  May need to tweak speeds but it almost sounds like the carriage is slamming into the sensor but when I look, there is clearance.  Could just be the speed and my perception but the disconnect happens when there is at least 100mm of travel to get to home.  

My SLB is mounted to the left front of the Y axis just in case that matters.

kglovern, 2024-04-09:
Do you get any feedback from the controller or software when this occurs?  If you're unable to see the console on disconnect, check the Safety area (Preferences -> safety) for any alarms or errors.  You can also check the diagnostic file console area (on disconnect, go to Calibrate -> Download Now) and check the console section to see if text was sent with further information.

Knowing what you have $22 set to would also help.

This is leaning a bit into the hardware territory so it might need to wait until Chris returns to go into depth a bit more there.

hamanjam, 2024-04-09:
Getting an alarm 8 when it disconnects.  [image]

hamanjam, 2024-04-09:
All good on waiting until Chris has had a chance to sleep.

kglovern, 2024-04-09:
OK, two things to try

1)  Change the debounce rate on the homing switches ($26) to a higher value.  If you're sure you see clearance, this might be the core issue.  I'd start at 250ms and tweak down (if it fixes it) until you find a balance between reliability and homing speed.

2)   If debounce doesn't help, try increasing pulloff distance ($27) to 2mm to start.

Past that, any further info/video might be helpful for Chris.

hamanjam, 2024-04-09:
Way more than you ever wanted to know.  I have a video but can't upload here I guess.  I'll try bounce settings and see where I get.  Everything was at the default.

[gSender-firmware-settings-4_9_2024-2_30_55 PM.json](https://github.com/Sienci-Labs/gsender/files/14923109/gSender-firmware-settings-4_9_2024-2_30_55.PM.json)

hamanjam, 2024-04-10:
Looks like setting $26 to 275+ solves the disconnects.    I'm assuming the disconnect is a safety byproduct of movement errors

closed 2024-04-10 by hamanjam