Sienci-Labs/gsender #501: Cannot use 4 corner "jog" buttons

hamanjam, 2024-04-10:
Getting the attached error when trying to go to any of the 4 corner presets.  I can manually walk to each of the 4 corners with the directional job buttons but not the quick corners.    No console message, only the message in lower left corner and no movement.  Homing on, inductive sensors working properly.  I've reset back to factory defaults and set up my firmware settings that I can find for all options I'm using.

[image]

hamanjam, 2024-04-10:
gSender 1.4.6 with SLB

kglovern, 2024-04-10:
Last minute enabling of the A-axis by default changed some of the mask values that will need to be accounted for.

While we'll long term need to fix this in the next release, in the meantime you can turn **off** the A-axis homing invert on $23 that's enabled by default.  This should set the mask back to a value that'll work with the function, and we'll fix the function itself for next release.

[image]

hamanjam, 2024-04-10:
That took care of the error.  It also fixed a confusing issue where the diag tab showed my homing position in back right where it should be front left.  I thought they were all related so I was looking at stepper wiring and all the inverse settings.

Appreciate the replies.  I'm not trying to be a pain, just pointing out things as I come across them since the boards are shipping and there's about to be a flood of new users

closed 2024-04-10 by hamanjam