Sienci-Labs/gsender #503: No toolchange prompt

gcormier, 2024-04-14:
1.4.6, Windows 11

I've done toolchanges before, not sure what has happened.

gsender sees the tools

[image]

```
...
G1X35.135Y-128.676
G1X34.196Y-128.593
G0Z6.350
G0Z25.400
M5
M6 T150
S24000M3
G4P6
G0X15.848Y-110.007
G0Z6.350
...
```

You can see we are at the last XYZ before toolchange.

[image]oBut it's just Resume Job or Stop Job as an option.

When I ran the file, it did the proper toolchange sequence after hitting Start Job (probe initial, probe new, everything looks good).

gcormier, 2024-04-14:
Config

[image]

gcormier, 2024-04-14:
[test_toolchange.txt](https://github.com/Sienci-Labs/gsender/files/14971844/test_toolchange.txt)

This file will repro the issue for me.

kglovern, 2024-04-16:
Thanks, we'll take a look.

kglovern, 2024-05-08:
This should be fixed in the next release.

closed 2024-05-08 by kglovern

gcormier, 2024-05-09:
Any guesses which version this bug appeared in? Have a few multi tool jobs to run this weekend. I can downgrade until the next version. (Or is EDGE a nightly build?)

gcormier, 2024-05-11:
1.4.4 seems OK