Sienci-Labs/gsender #507: Enhancement - move errors and job history to standalone files

hamanjam, 2024-04-30:
I would like to be able to clear the error history occasionally and maybe periodically clean job history.    Sometimes it gets clouded when trying to troubleshoot errors to have so much in the error log and I don't see a way to reset.  Right now, it appears to be the sender_rc file that holds everything and that can get big after a while.  Would be a "nice to have" enhancement for down the road.

hamanjam, 2024-11-29:
The ability to clear errors is there which is nice but it seems that the larger the history most gets, the slower gSender is to start up.  I have over 900 jobs in the history list (9 pages x 100 entries per page).

kglovern, 2026-09-15:
This feature is part of 1.7.0-Edge, so I'll be closing this as complete.  Thanks for reporting!

closed 2026-09-15 by kglovern