Sienci-Labs/gsender #508: Docker build is failing

Melkaz, 2024-04-30:
Hello,

When attempting to build the container image, I get this error:

```
[+] Building 11.8s (8/12)                                                                                                        docker:default
 => [internal] load build definition from Dockerfile                                                                                       0.0s
 => => transferring dockerfile: 994B                                                                                                       0.0s
 => [internal] load metadata for docker.io/library/debian:bullseye                                                                         0.4s
 => [internal] load .dockerignore                                                                                                          0.0s
 => => transferring context: 2B                                                                                                            0.0s
 => [1/8] FROM docker.io/library/debian:bullseye@sha256:f860cec7ccaf31b42cd60b6a409f9dad9510ea27fca9c2864741087b4298a1e3                   0.0s
 => [internal] load build context                                                                                                          0.0s
 => => transferring context: 34B                                                                                                           0.0s
 => CACHED [2/8] RUN apt-get update -y && apt-get install -y -q --no-install-recommends   apt-utils   build-essential   ca-certificates    0.0s
 => CACHED [3/8] RUN git clone https://github.com/nvm-sh/nvm.git "/root/.nvm"   && cd "/root/.nvm"   && git checkout `git describe --abbr  0.0s
 => ERROR [4/8] RUN npm install -g npm@latest && npm install -g yarn && yarn --production                                                 11.3s
------                                                                                                                                          
 > [4/8] RUN npm install -g npm@latest && npm install -g yarn && yarn --production:                                                             
11.08 /root/.nvm/versions/node/v14.17.6/bin/npm -> /root/.nvm/versions/node/v14.17.6/lib/node_modules/npm/bin/npm-cli.js                        
11.08 /root/.nvm/versions/node/v14.17.6/bin/npx -> /root/.nvm/versions/node/v14.17.6/lib/node_modules/npm/bin/npx-cli.js                        
11.15 npm WARN notsup Unsupported engine for npm@10.6.0: wanted: {"node":"^18.17.0 || >=20.5.0"} (current: {"node":"14.17.6","npm":"6.14.15"})  
11.15 npm WARN notsup Not compatible with your version of node/npm: npm@10.6.0                                                                  
11.16 npm WARN notsup Unsupported engine for libnpmexec@7.0.10: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.16 npm WARN notsup Not compatible with your version of node/npm: libnpmexec@7.0.10
11.16 npm WARN npm@10.6.0 had bundled packages that do not match the required version(s). They have been replaced with non-bundled versions.
11.17 npm WARN notsup Unsupported engine for libnpmversion@5.0.2: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.17 npm WARN notsup Not compatible with your version of node/npm: libnpmversion@5.0.2
11.17 npm WARN notsup Unsupported engine for libnpmpack@6.0.9: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.17 npm WARN notsup Not compatible with your version of node/npm: libnpmpack@6.0.9
11.17 npm WARN notsup Unsupported engine for pacote@17.0.7: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.17 npm WARN notsup Not compatible with your version of node/npm: pacote@17.0.7
11.18 npm WARN notsup Unsupported engine for @npmcli/run-script@7.0.4: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.18 npm WARN notsup Not compatible with your version of node/npm: @npmcli/run-script@7.0.4
11.18 npm WARN notsup Unsupported engine for read-package-json@7.0.0: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.18 npm WARN notsup Not compatible with your version of node/npm: read-package-json@7.0.0
11.18 npm WARN notsup Unsupported engine for @npmcli/run-script@7.0.4: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.18 npm WARN notsup Not compatible with your version of node/npm: @npmcli/run-script@7.0.4
11.19 npm WARN notsup Unsupported engine for pacote@17.0.7: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.19 npm WARN notsup Not compatible with your version of node/npm: pacote@17.0.7
11.19 npm WARN notsup Unsupported engine for @npmcli/run-script@7.0.4: wanted: {"node":"^16.14.0 || >=18.0.0"} (current: {"node":"14.17.6","npm":"6.14.15"})
11.19 npm WARN notsup Not compatible with your version of node/npm: @npmcli/run-script@7.0.4
11.19 
11.19 + npm@10.6.0
11.19 added 103 packages from 25 contributors, removed 315 packages and updated 122 packages in 10.295s
11.25 internal/modules/cjs/loader.js:892
11.25   throw err;
11.25   ^
11.25 
11.25 Error: Cannot find module 'node:path'
11.25 Require stack:
11.25 - /root/.nvm/versions/node/v14.17.6/lib/node_modules/npm/lib/cli.js
11.25 - /root/.nvm/versions/node/v14.17.6/lib/node_modules/npm/bin/npm-cli.js
11.25     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:889:15)
11.25     at Function.Module._load (internal/modules/cjs/loader.js:745:27)
11.25     at Module.require (internal/modules/cjs/loader.js:961:19)
11.25     at require (internal/modules/cjs/helpers.js:92:18)
11.25     at Object.<anonymous> (/root/.nvm/versions/node/v14.17.6/lib/node_modules/npm/lib/cli.js:2:18)
11.25     at Module._compile (internal/modules/cjs/loader.js:1072:14)
11.25     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1101:10)
11.25     at Module.load (internal/modules/cjs/loader.js:937:32)
11.25     at Function.Module._load (internal/modules/cjs/loader.js:778:12)
11.25     at Module.require (internal/modules/cjs/loader.js:961:19) {
11.25   code: 'MODULE_NOT_FOUND',
11.25   requireStack: [
11.25     '/root/.nvm/versions/node/v14.17.6/lib/node_modules/npm/lib/cli.js',
11.25     '/root/.nvm/versions/node/v14.17.6/lib/node_modules/npm/bin/npm-cli.js'
11.25   ]
11.25 }
------
Dockerfile:29
--------------------
  27 |       && nvm use --delete-prefix default
  28 |     
  29 | >>> RUN npm install -g npm@latest && npm install -g yarn && yarn --production
  30 |     
  31 |     WORKDIR /gsender
--------------------
ERROR: failed to solve: process "/bin/sh -c npm install -g npm@latest && npm install -g yarn && yarn --production" did not complete successfully: exit code: 1
```

It doesn't look like an error related to my architecture or docker version but here are the info: 
- Intel x86-64
- Docker version 26.1.0

I'm surprised to get an error building the container-image from the Master branch, I may have done something wrong :smile: 

Thanks !

J-eremy, 2024-05-01:
The Dockerfile is not buildable/functional. Its a placeholder by the looks of it. Currently the entire build process leans heavily on electron and has no direct way of getting it running stand alone without being really hacky. I imagine that in the future given time availability they plan on touching upon that but right now there are no options for docker unless you write it yourself.

If you look at the end of the file, you can see it is just an exercise in installing dependencies and listing the files in the directory and not actually doing anything productive. There isn't even an entry point defined.
[image]

I wrote a Dockerfile not too long ago to do the whole process but I wouldn't call it production ready, as it gets pretty complicated with the build/run phase and they have stuff like sentry.io running that I don't want to pollute their crash report infrastructure with irrelevant information.

kglovern, 2024-05-02:
Running the dev build/script is very easy if you're looking to get started and are used to node projects.

closed 2024-05-08 by kglovern