Sienci-Labs/gsender #512: stalled job

tcurdt, 2024-05-19:
I am running Version 1.4.7 on macOS talking to grblHAL.

I am running a job. I can see movements being buffered (different color in the visualizer) but at some point this stops.
The CNC then catches up and the job just stays idle doing nothing.
I repeated this a couple of times with always the same result.

I then opened UGS and it run the very same gcode file without any problems.

SienciLabs, 2024-11-05:
@tcurdt would you be able to attach a sample file so we can check on our end? Also it would be good to get the Diagnostic PDF so we can see what your machine buffer is set to. If the issue is now resolved on current version 1.4.10 then that would also be good news :)

tcurdt, 2024-11-05:
Sorry, I reported the issue in May. I fear I don't have the file anymore 🙃

I've encountered it again though. And fair enough, repeating a few commands on the console got the job going again - for a while at least. It sure is a buffering issue and the last version I have encountered it was 1.4.9. I can test 1.4.10+ in December.

tcurdt, 2024-12-16:
It's still happening in 1.4.11. I have attached the gcode. But I cannot always reproduce it.

[foam-big.nc.zip](https://github.com/user-attachments/files/18158113/foam-big.nc.zip)

tcurdt, 2024-12-17:
Last cut was terrible. I had to force it from idle probably 60 times to get the project finished.
Forcing works by sending `$$` from the console.

The visualizer shows some really weird behaviour. I would guess the buffer should always be ahead of the cutting operation. But that's not always the case. The buffer runs empty and sometimes it has to even catch up to the tool, to then overtake it to be ahead again.

I don't know if that's just the visualization, but it seems like the buffer handling has some serious problems.
The problems seem intensified when there is a lot of gcode for little movement. (non-arc roundings for example).

tcurdt, 2024-12-20:
After some more research and painful experience I now believe that these two issues are related.

https://github.com/Sienci-Labs/gsender/issues/513

I have made video.

https://github.com/user-attachments/assets/202519c3-2558-4a7d-8046-1afcc0415512

As you can see for every`$$` the number of lines counts down - and only then the job finishes.
It somehow always falls back to "idle".

tcurdt, 2024-12-28:
@SienciLabs Anything I should test before I leave the machine tomorrow?

Something feels very broken here.

SienciLabs, 2025-02-26:
@tcurdt there is definitely something going wrong here, thanks for the videos and files. I think our plan of action will be to look into testing if this same issue exists in gSender Edge and then solving it in Edge if it still does exist. 

The plan right now is to have Edge replace Main sometime in April and then continue forward with fixes and improvements with that new code base rather than straining our team in maintaining two separate code bases at the same time. Sorry to put your issue on hold a bit longer, but I really appreciate your reporting in. I didn't get back to you in Dec because I was seeing my family for Christmas.

SienciLabs, 2025-07-16:
@tcurdt I just realized I misread your comment as you identifying that this issue still exists in 1.5.#, is this true that you're still experiencing this in 1.5.# ?

tcurdt, 2025-07-16:
@SienciLabs TBH it's been a while.
I remember I also tried some Edge builds but cannot remember which version they were.

I can give an update on this in about 2-3 weeks with 1.5.x.
I will be testing on macOS (arm) and on Debian (on RPi).

SienciLabs, 2025-08-27:
@tcurdt if you have any more updates on this then let us know since we could spend more time looking into it. We looked more into it for 1.5.4 and implemented a couple other changes to avoid the issue

tcurdt, 2025-08-27:
Testing was blocked by https://github.com/Sienci-Labs/gsender/issues/703 ... next time when I am at the machine.
Which unfortunately might take a while.

SienciLabs, 2025-08-27:
Sorry to have delayed you, totally understand