Sienci-Labs/gsender #513: does not end job

tcurdt, 2024-05-19:
I am running Version 1.4.7 on macOS talking to grblHAL.

At the end of the job the job does not really end.

[image]

I've attached the gcode and do not see where anything is wrong.

[plate-2-CutRounds-2.nc.zip](https://github.com/Sienci-Labs/gsender/files/15370782/plate-2-CutRounds-2.nc.zip)

SienciLabs, 2024-11-05:
@tcurdt has this issue continued into the current 1.4.10 version?

tcurdt, 2024-11-05:
I have also seen it in 1.4.9 and haven't tested 1.4.10 yet.

SienciLabs, 2024-11-06:
Ok, let me know what you see. I ran your file on grbl and grblHAL boards and both times it completed - there was only a short delay before it marked as completed at the end of the job

tcurdt, 2024-12-16:
Just tested in 1.4.11 and took a screenshot 

[image]

tcurdt, 2025-07-11:
> I'm very impressed how reliably you've been able to make this happen meanwhile I still wasn't able to reproduce it, but that's something I can keep working on on my end if needed

Challenge accepted 😅

I'll be with the machine in about 2-3 weeks and will then report back.

walidkayhan, 2025-08-08:
Just wanted to check in on this, are jobs that should be done but aren't still a thing for you @tcurdt on the latest version of gSender, v1.5.3?

tcurdt, 2025-08-08:
I was just trying today but had to give up for the day and get some fresh air instead.

https://github.com/Sienci-Labs/gsender/issues/703

The upgrade killed my working config. 😔

Will try tomorrow or on Sunday again.

tcurdt, 2025-08-28:
Nice. Will take a while until I can test it but I will let you know.

tcurdt, 2026-01-09:
I was running gsender edge over xmas. It **seems** to be fixed.
Carefully closing and fingers crossed.