Sienci-Labs/gsender #516: Feature Request

CybergNetwork, 2024-06-01:
Resume Job if there is a power outage. Same as found in 3D printers. Just a scripts that reads the gcode and remember where it was at. Even if was starting from a few lines above the line it was working on, it would be a time & resource saver.

Thank you and keep up the good work!

kglovern, 2024-06-03:
We actually have start-from-line functionality already built into the software - documentation can be found here:

https://resources.sienci.com/view/gs-using-gsender/#job-loss-recovery

CybergNetwork, 2024-06-03:
You guys thought about everything! lol Thank you for educating me. You can delete my post then.

Have a great day.

https://resources.sienci.com/view/gs-using-gsender/#job-loss-recovery

closed 2024-06-03 by kglovern