Sienci-Labs/gsender #518: Fixed Tool Length Sensor not working

ShadyMCShadesen, 2024-06-06:
Hey, the wizard for the fixed tool length sensor is not working properly, i'll try to explain the weird behavior:

### **First of all, here are my settings for the FTLS:**
[image]

### **Everything is explained on the screenshots of the console logs:**
[image]
[image]

Im running gSender in the version 1.4.7

mattydboom, 2024-09-18:
I am running 1.4.8 and get the same weird issue where; on some occasions the Tool change get called up twice. Despite successfully completing the first tool change probe etc. it then goes to start a the cut and then starts another tool change. There looks to be an additional 'M06' command that gets added into the gcode and calls for 'T-' which isn't in the raw gcode file???

SienciLabs, 2025-07-22:
@ShadyMCShadesen and @mattydboom we've made several updates to tool change handling since 1.4.7 and 1.4.8, can you update to 1.5.2 and let us know if you're still having issues?

walidkayhan, 2025-08-11:
@ShadyMCShadesen are you still experiencing this bug on v1.5.3?

walidkayhan, 2025-09-19:
@ShadyMCShadesen just following up on this. Has this issue been resolved for you in version 1.5.4?