Sienci-Labs/gsender #522: Stuttering movement using game controller

uzadow, 2024-06-17:
I get stuttering/jerking movement when using a game controller that essentially makes jogging unusable.

Setup: Raspberry Pi, gSender 1.4.7, noname ('shanwan') USB playstation controller.

This worked with a gSender 1.2.2 install - updating to a current Pi OS and gSender broke it.

Any help debugging this would be appreciated, and I'm happy to provide more information if needed :-). Also happy to provide a video of the symptoms if that helps.

Ezekiel-DA, 2024-06-22:
Seconding this, but on Windows. Also gSender 1.4.7. Have not tried previous version (this is what I started with since I just got my Longmill MK2 a month or so back), happy to try earlier version is that is helpful for debugging.

uzadow, 2024-06-23:
Here's a video of the stepper motor movement - hope that helps.

Downgrading to 1.2.2 fixed this on my end. Still happy to help debug this :-).

https://github.com/Sienci-Labs/gsender/assets/7524857/4d497cf5-6749-43e9-acf1-9ee4c00eb048

hamanjam, 2024-06-24:
My Logitech wireless F710 does the same thing. Keyboard ardor shortcuts are fine, just the gamepad stutters bit I believe that has happened in the 1.4.x release.  I rarely use the gamepad so it wasn't critical for me to look into anything

walidkayhan, 2024-07-09:
For versions 1.4.x, we drastically changed how jogging works with gamepads on gSender. One of the new features we added is dynamic jogging, where the speed is dependant on how far you push the thumb sticks on your controller. This can lead to a stuttering affect as we aren't able to send a single long jog command like we did before, but instead have to send many small jog movements one after another.

If you want to eliminate the stuttering, you can play around with the movement distance override setting in your gamepad profile within the settings.

[image]

Some users reported setting it to 120% or 150% helped get rid of the jog stuttering on their end but you would need to experiment with it to see which value works best for you :)

uzadow, 2024-07-12:
Thanks for your support!

I've tried this now. The stuttering is less at 150% and mostly (but not completely) disappears at 500%. But now, there is an additional issue: If I jog in one direction (e.g., left), stop, and then move in the opposite direction (e.g., right), the movement is sometimes left, stop, then left again, then right.

I've reverted to the old SD card again,  but I'm happy to do more tests if it helps.

lappa1, 2024-08-18:
Could we please have this dynamic jogging as an option?
I personally like to have very accurate control when jogging which is very hard to achieve with dynamic jogging.

mattydboom, 2024-08-20:
Experienced the same stuttering issue at the default 100% override setting, and upping to 125, 150, 300 all made only marginal improvements. Wasn't until I hit 500% as suggested by @uzadow above that things started moving much more smoothly. You can hear when each jog gets initiated still, but is a vast improvement over the lower settings. 

I also haven't noticed any movement in an incorrect direction afterwards either so would say this is a sufficient fix (for me) for now, so wanted to share in case it could help anyone else looking here..... I would never have thought to touch that setting, let alone bump it up so high, and as such think think this could be better handled in the UI.

uzadow, 2025-07-19:
Thanks for the update!

I tried 1.5.1 on my setup, and unfortunately, the issues persist. At 100%, I get the same stuttering. At 500%, I get somewhat consistent but still jerky movement.

I'd be happy to do more testing, and also happy to do small code/config changes if you can point me in the right direction :-).

walidkayhan, 2025-08-08:
I've added to our backlog, adding a feature to choose dynamic or standard jogging for controller thumb-stick movements. We haven't added it to our sprint plans yet, but we will get to it at some point. For all those who have been vocal about wanting that feature, be on the lookout for new versions of gSender in the future :)

walidkayhan, 2026-02-03:
This has been added to the [1.6.0-Edge-3](https://github.com/Sienci-Labs/gsender/releases/tag/v1.6.0-Edge-3) release, thanks to @kesmit13 who opened a pull request. Eventually the changes in the edge releases will be merged into a stable release. Will close this issue for now but feel free to open a new one if you find any issues with this feature :)