Sienci-Labs/gsender #525: Y axis problem

philhape, 2024-07-06:
System locks up when I jog the Y axis. X and Z work fine. Y axis works fine when I run a file. Only problem is when I try to jog the Y axis

mattydboom, 2024-08-03:
I get tis too but only when using the corner homing buttons shown below
[image]

Everything is set correctly and runs fine otherwise (Homes fine with the normal $H Command etc.). 

I think I know where the error is though.... seems to be linked to the fact the Y-Axis is reversed on my setup, and the command line generated for these buttons gives a negative direction when it should be positive.

[image]

The result being that the soft limits are (correctly) being triggered by the incorrect value. 

Put anther way, I think the coding hasn't taken into account the limit switches being on the negative end of the Y-Axis travel.

Breaker911chhh, 2024-08-11:
I have the same problem, but for me the issue only occurs when I try and continuously jog the machine in the y axis. I can single click to move in the y axis and it works just fine. The only time it "locks" up is when I click and hold to jog in the y axis. X and Z axes work both with single click or click and hold.

SienciLabs, 2024-11-05:
@philhape @mattydboom @Breaker911chhh how are these issues for you now with recent 1.4.10 version of gSender?

mattydboom, 2024-11-05:
@SienciLabs just checked, and all these buttons are still throwing up alarms when pushed. It seems the directions requested in the gcode commands are all wrong..... and so triggering the soft limits. The values seems OK just the fact they have incorrect + or - values.

Perhaps related to the fact the button containing the 'Home' symbol is shown as top right, where my machine actually homes to bottom left (as below). It is all set up this way in the FluidNC config.yaml file so not sue why this doesn't get pulled in correctly..... but if this can be altered/fixed then perhaps the other homing buttons wouldn't trigger the soft limits

[image]

mattydboom, 2024-11-05:
@SienciLabs to assist with the above I am attaching my config file as while I appreciate it is not for your controller offering, I figured it may help you to identify / understand the problem. I have had to change to .txt format to allow upload. Let me know if I can be of any other help.

[config.yaml.txt](https://github.com/user-attachments/files/17638804/config.yaml.txt)

SienciLabs, 2024-11-06:
Ok thanks @mattydboom. In your situation I can see why this is causing issues, since we still don't offer official support for FluidNC so gSender has no ability to read the yaml and determine your home location. Because of this, it's defaulting to the top right corner for homing and then the corner button values are getting the wrong sign as you said.

In this particular case it might not be something that's resolved anytime soon because we still have such a large backlog of work to do and feature requests for controllers we already support. Perhaps if you or anyone else with some coding chops would be able to contribute to the code base with a new FluidNC 'controller' (there's infrastructure we've implemented in gSender to handle behaviour differently based on the controller a.k.a firmware) then this issue for your machine would be able to be more fast-tracked. The file you attached is useful for reference in either case so thanks for responding 🙏

mattydboom, 2024-11-07:
@SienciLabs ah I see, and good to know the reason. I would love to see FluidNC supported as I know it has become a very popular firmware for DIY hobbiyst machines. I fear my coding skills are far from up to scratch however, so will see if anyone on their discord forum that uses GSender is willing to input.

**That said** - could it be a simple option under the firmware to include a 'FluidNC' or 'generic' controller option that once selected, prompts you to pick the homing location. I note that most (if not all) other homing related information is dragged into the firmware section so with this last bit of information by user input, all homing buttons could work as intended

SienciLabs, 2024-11-07:
Interesting idea - though since there's also been other feedback on FluidNC support then perhaps community-built support for it might not be too far off. Either way I'll keep the thread open for now and your idea in mind