Sienci-Labs/gsender #527: Go XY0 raises z but doesn't lower.

Plutoaurus, 2024-07-22:
New convert to the software.

When clicking the "Go XY0" button, it raises z, moves but then doesn't lower again.

The problem comes when I'm moving around to check boundaries etc and go back to 0 the z crashes into the upper limit.

Is there anyway to change this?

Thanks!

SienciLabs, 2024-11-05:
Right now we don't have a way to change this. We discussed for a while how we wanted to handle Gotos since of course we can always make a direct movement, but then we got some feedback that it'd be good to move at a safe height so we added that, but then we were uncomfortable to make it so the Z-axis lowers again by the safe height at the end of the movement in case something is there that the CNC will crash into

SienciLabs, 2025-07-17:
Coming back to this I think I'll close this. Without knowledge of whether the Go To might run into something when it moves back I think most people wouldn't want the bit moving back down out of caution. Luckily the safe height is a value you can set in the settings to decide the value of the initial move upwards so you can always make this zero which would technically remove this problem for you.

Thanks for submitting the original feedback!