Sienci-Labs/gsender #532: Outline not behaving as expected

mattydboom, 2024-08-09:
I have found that the Outline trace function does not generate the desired path for some projects. See below example. The 'expected' path shown in Green, but the actual generated outline path is something more like that shown in Red.

I suspect this an issue with the circular tool path island, but as this is not uncommon of a typical job & otherwise is just a simple rectangular shape, I would expect it to be dealt with correctly. I am also attaching the tool path code to aid with the problem finding.

[image]

[image]

[Outline Issue.txt](https://github.com/user-attachments/files/16567040/Outline.Issue.txt)

SienciLabs, 2024-11-06:
@kglovern maybe we could introduce a new toggle in the settings to switch between a more simple and a more complex outline? Since I know where we run a simple box we didn't tend to have these issues, but since we switched to the more complex form outlining sequence there are sometimes exceptions that appear. I think people would appreciate the more complex one more, but if it's making odd movements like this then that'd give the simple one to fall back onto. Or maybe there's some other way we could introduce some type of error checking in the outline feature to avoid situations where it collapses in on itself like this

mattydboom, 2024-11-07:
@SienciLabs @kglovern yes I think having the option would be a great idea. 

I personally use the outline function to quickly get an idea of the extents of the project (to make sure its entirely sitting within the stock)..... so just having a bounding square / circular outlining path would be great for this. 

A bit like the below that is available within Lightburn. Perhaps on the new UI the outline tool can have a drop down allowing you to change between the three modes (Complex, Square, Circular) and will always remember/stay on your last selected choice.

[image]

SienciLabs, 2025-07-22:
I'm bumping this as something we'd still like to implement as a good, temporary solution

walidkayhan, 2025-08-08:
@mattydboom We recently released v1.5.3, which includes some improvements to outline, as well as being able to choose the style of outline you would like.

[image]

Would you be able to update to the latest version of gSender, try out outline, and let me know if this is more so what you are expecting from the outline feature? Appreciate it :)

mattydboom, 2025-08-08:
@walidkayhan thanks, and yes - I spotted that beauty of a release note and almost spat my tea out with excitement. I have already run several jobs using this feature.... and although I quite literally LOVE it, I have noted an oddity with the way it outlines and have been meaning to mark-up some screen shots to explain the behaviour. I will post back here very soon to explain.

mattydboom, 2025-08-09:
@walidkayhan as mentioned. The issue with Square outline as it currently stands, looks to be that it assumes your toolpath is located at the 0,0 origin. To demonstrate, I have set up a simple 300mm square toolpath and offset it from the origin by 100mm in both X&Y. 

The **expected** behaviour would be for the square outline to track to the nearest corner and then follow the perimeter of the square (see below).
[image]

The **actual** current behaviour is to follow the X&Y zero axis before crossing to/from the squares perimeter at the sides furthest from the origin.
[image]

I cant personally see any advantage to this so would have thought an adjustment to this feature is need to become fully realised - but looks forward to hearing your thoughts on this, and thanks again for taking on board my (and other users) feedback and implementing!

mattydboom, 2025-08-12:
@SienciLabs oh and yes I have tried the Detailed outline out too. 

It does seem to track a lot faster like you mention. Although with circular / curved tool path edges in particular, I find it to be a touch erratic, and for a perfect circle (like in my test example above) it doesn't seem to follow the perimeter perfectly...... making some straight edges here and there. I suspect this is contributing to the jolty nature of this trac method.

Handy when you are trying to use up a odd shaped / partially used piece of material... but for me the square outline has more utility in everyday job setup. I appreciate its somewhat horses for courses though!

It also seems to trace the travel moves to and from the origin at the start/end of its trace, so if it were possible to ignore this element & somehow fix for both outlining methods, I think that would be a good result & everyone should be happy :)

closed 2026-03-02 by kglovern