Sienci-Labs/gsender #541: Suggestion: Dialog boxes in macros

AndyCXL, 2024-08-31:
To enhance macro functionality and safety, it would be really neat if simple dialogs could be popped up asking for confirmation of some fact or position - with the option to Cancel or Proceed.
Were this an accepted line of thinking to the developers, allowing dialogs to feed back a choice/choices and perhaps a value/values would open up a whole new world for macro creators…

J-eremy, 2024-09-02:
> To enhance macro functionality and safety, it would be really neat if simple dialogs could be popped up asking for confirmation of some fact or position - with the option to Cancel or Proceed. Were [...]

You can already somewhat do this with the M0 command. If in your macro you have a line like this
`M0 ;This is a test`

It will pause whatever you are doing and display this dialog box
[image]

The ";This is a test" part of it is a comment, notice the semicolon which defines the proceeding text as a comment, and is what is displayed in the pop-up.

Check out the "Advanced Macros" Section of the following URL.
https://resources.sienci.com/view/gs-additional-features/

_Advanced Macros
gSenders Macro architecture is based on JavaScript and uses the Esprima library (https://esprima.org/) and so will theoretically support any code that it does. This is exciting because Macros can move far past basic variables if you’d like to perform advanced functions on your CNC:_

There is also a ton of built-in macro variables for things like tool number and positional macros.

SienciLabs, 2024-09-05:
Good point @J-eremy, I realized I hadn't explicitly mentioned it in the documentation so your comment here reminded me to add it in there https://resources.sienci.com/view/gs-additional-features/#advanced-macros

@AndyCXL it sounds like you were looking for functionality that went even further?

AndyCXL, 2024-09-05:
The ability as pointed out of using "M0 ; Brief instruction as a comment" certainly addresses an important part of my suggestion, but there are instances where gaining an input from the user via a (modal) dialog popped up during a macro run would be useful. 
An example might be probing a horizontal cylinder to find the Z-centre... if the user probes something 30mm dia or say the chuck on a Vortex at 100mm dia, the macro could behave 'smarter' and avoid time wasted whilst a G38 command seeks first probe contact. Equally, if the macro is configured to only accommodate a 30mm cylinder, but the user unthinkingly tries to probe the chuck, it could result in a probe crash.
Here, asking the user for the approx dia of the cylinder before moving would be very useful.