Sienci-Labs/gsender #545: No Continuous Jogging via Keyboard

slynn1324, 2024-09-09:
I'm not sure if this is expected behavior, but from what I can tell in the code and documentation it is not.   I can't get continuous jogging to work via the keyboard shortcuts.  I get individual step moves, but holding the key combination (e.g., shift+right_arrow), it won't start a continuous jog.  When using the jog controls via the mouse on the UI, continuous jogging works as expected.  I also tried to change the shortcut a single key, but there was no change. 

I am running Linux x64 - Debian 12 with XFCE.

slynn1324, 2024-09-11:
I did some more digging here, and while I'm pretty unfamiliar with the code overall I was able to get some additional logging added and found what may be the culprit.  

I found that by the time the jog command gets into the GrblController.js "jog:start" handler, the axis character received when using the keyboard shortcuts is lowercase 'x', where the axis character when jogging from the UI button is uppercase 'X'.  Looking further below in this handler, all of the math appears to only look for the uppercase versions of the axis characters.   By adding a bit of code here to convert 'x' to 'X', 'y' to 'Y', and 'z' to 'Z', the keyboard shortcuts appear to be working as I would expect.  

There is almost certainly a better place and way to address this so I'm not submitting a proper PR at this time.  I'd be happy to help more where I can to get this into a proper fix release, but will need some guidance. 

`@@ -1886,6 +1887,22 @@ class GrblController {
                 let unitModal = (units === METRIC_UNITS) ? 'G21' : 'G20';
                 let { $20, $130, $131, $132, $23, $13 } = this.settings.settings;
 
+                //SL
+                log.debug('axes=' + JSON.stringify(axes) + ' $20=' + $20);
+                if (axes.x) {
+                    axes.X = axes.x;
+                    delete axes.x;
+                }
+                if (axes.y) {
+                    axes.Y = axes.y;
+                    delete axes.y;
+                }
+                if (axes.z) {
+                    axes.Z = axes.z;
+                    delete axes.z;
+                }
+
+`

kglovern, 2025-06-17:
Should have been addressed in a past build with a contributor pull request.