Sienci-Labs/gsender #549: Enhancement request - Laser/Spindle switch applies firmware settings

hamanjam, 2024-09-16:
Over the weekend, I installed the Sienci Laser.  It took a bit to figure out the required firmware settings and it seems that when I install the laser, I need to update firmware $30 & $31 ($32 appears updated by the Laser/Spindle switch and then when I switch back to the spindle, I have to set back to Spindle and then manually go into firmware and set those settings back to my spindle RPM.  

Would it be possible to allow the Laser/Spindle switch in gSender update $30 & $31 as well?    You could have user values in the spindle/laser tab and when that switch it flipped, the correct settings would be applied.

hamanjam, 2024-09-16:
It appears this functionality is in the laser/spindle tab but it's not changing firmware as expected.  I have to manually update firmware when switching

SienciLabs, 2024-09-17:
Thanks for the issue submission @hamanjam! I just responded to your post on Facebook and I'm pretty confident that if you follow my suggestions this problem should be dealt with for your setup. Try it out and let us know if that's the case 👍

https://www.facebook.com/groups/mill.one/posts/1985776111893710/

hamanjam, 2024-09-17:
Thanks for the response Chris.  I still have to figure out the "reboot" scenario and I'm getting a lot of errors changing things around in testing but I will gather the steps to recreate and open an issue for that if there isn't already one:

[image]

closed 2024-09-17 by hamanjam