Sienci-Labs/gsender #552: Remote profiles not synching to devices

b3nnb, 2024-09-22:
I setup gsender edge to access from any other device on my network (Which works) but anytime I go on a new device the settings and macros always have to be imported.

Should it not automatically do that?

I'm running the very latest gsender for Linux.
The OS I've been able to replicate the issue is pop_os! And Ubuntu, both running the current stable releases.

J-eremy, 2024-10-04:
#486

SienciLabs, 2025-07-08:
Currently this is a behaviour with how we set up Remote mode to work, but do agree that it'd be much more streamlined if the settings could automatically sync somehow to the remote devices. We'll keep this in mind into the future if it'd be possible to implement

SienciLabs, 2025-07-08:
From @siganberg 

> 
> Currently, the remote connection settings in gSender are independent and require separate configurations for each remote device. This includes settings like Park location, Probe, Automation, Tool Changing, Macros, etc.
> 
> Expected Behavior:
> 
> Remote connections should use the settings of the host server by default. This eliminates the need for importing and exporting configurations every time a remote device is connected. The settings for the remote devices should be inherited directly from the host server.
> 
> All settings changes made from a remote device should automatically sync to the host server. This ensures that both the remote and host configurations stay in sync without requiring manual updates.
> 
> Steps to Reproduce:
> 
> Set up a host server with configurations (Park location, Probe, Automation, etc.).
> Connect a remote device and notice that its settings are separate.
> Modify a setting on the remote device.
> The change does not sync with the host server, leading to inconsistent configurations.