Sienci-Labs/gsender #557: Cannot compile

Windfisch, 2024-10-01:
When I try to compile this repo, the `yarn install` step fails with the following:

```
[5/5] Building fresh packages...
[12/16] ⠄ core-js
[2/16] ⠄ electron
[8/16] ⠄ cypress
[11/16] ⠄ usb
error /gsender/node_modules/@serialport/bindings: Command failed.
Exit code: 127
Command: prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
Arguments: 
Directory: /gsender/node_modules/@serialport/bindings
Output:

```

This happens with Arch Linux, Debian stable and alpine:latest.

Additionally, the Docker file does not work. Also, `npm install` instantly fails on latest `main` with `ERESOLVE`.

What's the expected build environment and how to build this software?

J-eremy, 2024-10-04:
Make sure your using node14. Also might be best to compile it in a dev environmet inside a docker container. 

The Dockerfile doest work because there isn't currently a way to run headless without hackery, the entire process is wrapped in an electron app.

I also know there are a number of prebuild and postbuild scripts that you have to run also.

J-eremy, 2024-10-04:
I realized I didn't give you much information. Here is a Dockerfile I just made that builds and exports the built files back into your local directory. 

### I use the --output method to get it to export the built files which needs **buildx** from docker which might not be installed by default on your system but should be in your repos of whatever you're using.

```Dockerfile
ARG version=v1.4.9

FROM node:18-bullseye AS build-stage

WORKDIR /build

# Clone the repository
RUN git clone https://github.com/Sienci-Labs/gsender.git . && \
    git checkout ${version}

# Install dependencies
RUN yarn prepare && yarn install

# Build the project
RUN yarn prebuild-latest && yarn lint && yarn build

FROM scratch AS artifact

#Copy the built files from the build stage to the --output directory
COPY --from=build-stage /build/dist /dist
COPY --from=build-stage /build/bin /bin

# This will dump the build and the bin directories out into the folder with the Dockerfile.
# You can then run these commands to get it running outside of a container
# After build, I tested it running on Node 21.6.2 as of writing this.
# yarn --cwd dist/gsender install
# npm install electron --prefix dist/gsender
# ./bin/gsender

```
This can be successfully built using this command:
`DOCKER_BUILDKIT=1 docker build --target artifact --output . .`

Result:
[image]

This will dump the build and the bin directories out into the folder with the Dockerfile.
You can then run these commands to get it running outside of a container
`yarn --cwd dist/gsender install`
`npm install electron --prefix dist/gsender`
`./bin/gsender`
**After build, I tested it running on Node 21.6.2 as of writing this. and it runs headless**
**NO AI WAS USED IN WRITING THIS**