Sienci-Labs/gsender #568: 1.49 Rotary Axis Keyboard Shortcuts not working

AndyCXL, 2024-10-25:
1.49 Rotary Axis Keyboard Shortcuts not working.

GrblHAL on Teensy4.1/Phil Barrett BOB/Modified ShapeOKO 3XL with XYZA permanent axes, dual-Y with auto-squaring, homing sensor on A. Fully functional and recently cut a rotary job, didn't for that job use the KB A+/A- shortcuts.

Tried stock shortcut settings, confirmed shortcuts are enabled.
Using KB shortcuts, no axis movement + or -
Click A+ and A- icons on rotary tab, axis moves correctly
Click Home A icon on home page, axis homes correctly

Set new shortcut keys, confirmed enabled.
Using KB shortcuts, no axis movement + or -

kglovern, 2024-12-16:
I was informed this was fixed in the latest version (1.4.11) - let us know if it's working for you.

AndyCXL, 2024-12-16:
I can confirm that this is now fully working, thank you

closed 2025-03-13 by kglovern