Sienci-Labs/gsender #570: v1.5.0-Edge-1 reports error loading library libs.so on ARM-64 for Linux

JohnOCFII, 2024-10-27:
Attempting to launch the AppImage `gSender-1.5.0-Edge-1-Linux-ARM-64Bit.AppImage` reports the following error:
`error while loading shared libraries: libs.so: cannot open shared object file: No such file or directory`

Debian ARM Linux:

Linux vmcnc 6.1.0-26-arm64 #1 SMP Debian 6.1.112-1 (2024-09-30) aarch64 GNU/Linux

osh@vmcnc:~/Apps$ ./gSender-1.5.0-Edge-1-Linux-ARM-64Bit.AppImage 
./gSender-1.5.0-Edge-1-Linux-ARM-64Bit.AppImage: error while loading shared libraries: libz.so: cannot open shared object file: No such file or directory
osh@vmcnc:~/Apps$

walidkayhan, 2025-07-07:
Can you confirm if this issue persists for you on the latest version, v.1.5.1, please :)

closed 2025-07-28 by SienciLabs

JohnOCFII, 2025-08-05:
Sorry for the delay.  That error no longer appears.