Sienci-Labs/gsender #575: FEATURE REQUEST: 1.5.0 EDGE - Always on screen macro/action bar separate form main function tabs.

J-eremy, 2024-11-07:
The new UI is looking great. I propose you add an action area so that things like the probing and macros can be used without swapping tabs.  A good place might be below the center area where feeds and speeds are modified as there is plenty of open space there.

A simple solution would be to split that main area into two, and have a row of square buttons that can be linked to the existing macros or probe commands. Like quick buttons to macros that are used a lot. This will also open up the ability to click a macro and also have the console open to monitor the action without trying to quickly switch tabs or have the console open as a separate window.

I think an always on screen shorcut bar would be beneficial for many things, not just my use cases.

I usually work on the bottom left of the machine, so the machine sits at work coordinates x-0 y-0 z-0, I also use a touch device to measure the initial bit and store that value, then if the job requires any other bits the new bits are measured and the offset is adjusted.

I also have a macro that brings the z height up to a height I can put the dust boot on after successful probe as after probe it only raises back up slightly.

Currently a common workflow I use is as follows:
1. **Bring spindle to convenient bit change area** ***
2. Install the first bit for job
3. **go to the center of stock I will be machining** ***
4.  probe z-height 
5. **send the machine off to measure the initial bit** ***
6. complete machining job for bit 1
7. **bring spindle to convenient bit change area if x0 y0 z0 isn't convenient.** ***
8. **send the machine to measure bit 2 and calculate offset**  ***
9. rinse and repeated for however many bits
10. **send the machine to a specific place I lovingly call GTFO so I can remove completed piece and clamp a new one down.** ***
11. Repeat as necessary.

The steps in bold with *** are all macros. I used to switch between macros and probing tabs, I bound "Start Probe" to a key bind and it drastically improved not only the speed but general user experience by not having to do the tab dance anymore at all.

With the new UI design being way more touch friendly, a set of customizable quick buttons would allow me to ditch the keyboard all-together.

[image]
This is an example of a single row, but 2 rows by my calculation would be roughly twice as useful +/-.

A good place to set them without going down a rabbit hole of new windows or popups would be in the already created key bind menu. You can just add another column to set which button that action is bound to on the main screen. That would simplify setting every action that is already bindable.

kglovern, 2026-09-15:
This will be part of 1.6.4 as part of your pull request - thanks for contributing!

closed 2026-09-15 by kglovern

J-eremy, 2026-09-16:
> This will be part of 1.6.4 as part of your pull request - thanks for contributing!

Its the least I could do. Thank you for making an awesome piece of software.