Sienci-Labs/gsender #579: Increase and decrease jog speed shortcut keys not working

DenEric, 2024-11-30:
Using mpcnc with Bigthree ocpotus controller. Increase and decrease jog speed shortcut keys are not working and destroy the "Rapid, Normal and Precise" settings.

kitizz, 2025-03-18:
I'm seeing the same thing while accessing via remote interface. As soon as I press `-` or `=`, the jogging stops working. But the GUI doesn't show any changes. But I change the Feed Present (Precise, Normal or Rapid), I see that the XY, Z and speed values are all numbers like `-0.6, -0.7 -0.87`.
If I try to change these back to their proper values, they reset to these weird values the moment I change the feed preset again.

walidkayhan, 2025-07-07:
We've made several bug fixes to keyboard shortcuts to gSender since then. Could you try out the latest version, v1.5.1, and see if these shortcuts are working now?

closed 2025-07-28 by SienciLabs

kitizz, 2025-09-10:
Confirmed it's working for me!