Sienci-Labs/gsender #582: Error1 on everything today

hamanjam, 2024-12-11:
I'm continously plagued with this Error 1 message but today, nothing is running.  Usually, I can restart gSender or reboot the computer but today nothing is working.  It doesnt' matter if there is a file loaded or not.  1.4.10 reduced the number of messages but today it is on everything.  My zeros are messed up and I can't complete any operation

[image]

hamanjam, 2024-12-12:
I may have found my trigger for this error.  I started a laser job and right away realized I forgot to turn on the laser drive so I stopped the job and now I can't do anything and the Error 1 pops up with any jog command I try or even goto xy0

kglovern, 2024-12-16:
Let us know if this is continuing on the latest version (1.4.11) - some changes were made to the line parser to make it play nicer with realtime commands.

hamanjam, 2024-12-21:
Still getting it randomly.  It usually occurs after some other error it stopping a job. 

Almost every time, if you use the goto icon and type in the XY coordinates but leave the ZA as the defaults (either z0 or A is some very high number). It will throw an error about being out of bounds then trying to quickly unlock results in the Error1. 

It's happened randomly one or two other times but far less than before so far so progress is good.

hamanjam, 2025-03-01:
I'll close this for now.  I just downloaded the new Edge version so I understand this isn't priority on 1.4.x

closed 2025-03-01 by hamanjam