Sienci-Labs/gsender #583: 1.5.0-Edge-2  No USB devices found

axg20202, 2024-12-13:
This is probably related to the new method of auto recognizing controllers. In the stable release, my controller is detected but not recognized - i have to manually select its COM port. No such ability in the Edge release, so can't connect to my controller.

kglovern, 2024-12-16:
We'll re-add the "unrecognized devices" area in the next edge release 👍

SienciLabs, 2025-02-26:
@axg20202 this should be available in the upcoming EDGE4, let us know if it starts working again for you. In the meantime, it'd be great if you could also follow this guide, and send back to us the hardware and vendor IDs for your CNC so we can eventually add it as a 'recognized device' https://resources.sienci.com/view/gs-feedback/#port-not-found

closed 2025-03-13 by kglovern