Sienci-Labs/gsender #584: FEATURE REQUEST : Biger coordinate system on visualizer.

macrumpel, 2024-12-13:
Hello,

on my screen the numbers of the coordinate system are very small and even impossible to read. It would be great to have this adjustable, as precision is important.
Thanks a lot !

Hubertus

kglovern, 2024-12-16:
Do you mean on the visualizer itself?

macrumpel, 2024-12-17:
Yes exactly. Just to verify that the object is in accordance to placement on the board. I use gSender for plotting, and it would help me to be sure that the drawing is on the paper.

macrumpel, 2024-12-20:
What would also be great is the ability to have the axis visible even if you zoom in. More like rulers actualle...

kglovern, 2026-09-15:
A large number of visualizer customizations including larger numbers, machine bed matching, and more are coming to 1.7.0-Edge so I'll be closing this as complete.

closed 2026-09-15 by kglovern