Sienci-Labs/gsender #585: GSENDER KEEPS SHUTTING DOWN

DRCARTER1763, 2024-12-15:
I just recently started having this issue  but for last week or so Gsender just shuts down in the middle of running a file. It leaves the spindle running the program closed. Same computer, same CNC, files are generated using Vcarve Desktop. Has me scratching my head.

Can you help me figure it out.

walidkayhan, 2025-01-06:
Which version of gSender are you using? Did you happen to update it recently?

DRCARTER1763, 2025-01-06:
I have tried all versions from 1.4.9 up to the latest version. All do
the same.I can't even use it anymore. I have a 1 year old HP 17" laptop
running windows 11 FYI.

Thanks,
Danny

hamanjam, 2025-01-07:
Haven you checked Windows Event Log?  There may be something in the windows applications log to say why the app is crashing

DRCARTER1763, 2025-01-07:
Hello,

So 2 things. First I don't know anything about Windows event logs or how to
use and read them. If you tell me exactly what to do I will do it and send
it to  you.
Secondly, Today I upgraded G Sender to version 1.4.11 and the problem still
persists maybe more frequently. I can hardly get through the probe zeroing
process without it shutting down.

I sure NEED you help BAD>

If you have a way to access my computer remotely, I will be happy to let
you do that.

Danny
910-305--0232

hamanjam, 2025-01-07:
For Event viewer, click start, and then type event and it should have an option for Event Viewer.  Open "Windows Logs" and click "Application" 

Scroll through the list for anything marked as Error.  This one is for MS-Teams but you may have something for gSender if Windows sees the crash.  It should hopefully have some indication why it's crashing

[image]

closed 2025-07-17 by SienciLabs

DRCARTER1763, 2025-07-17:
Good afternoon. I received the above replay to my voice mail left yesterday
which is totally unrelated to my current issue.
My current issue is on V1.5.1. It worked great for week or so and then all
of sudden when I open the application  it would not connect
to my cnc. GSender just  keeps on showing "connecting" but  never connects.
t
My other GRBL controller connects and runs my cnc fine. Also my Lightburn
will still connect and run my cnc with not issue/

Can  you please hilp me fix this issue?

Danny R. Carter
910-305-0232