Sienci-Labs/gsender #586: option to handle the A axis as a normal axis

fred-c1, 2024-12-15:
Hello,
Sometimes I use my CNC for 3D printing.
Most slicer softwares can generate gcode for CNCs (MACH3 or GRBL mega 5X): the extruder stepper motor is call "A" (instead of "E0").
It seems that gsender uses the "Y" axis as a rotary axis (instead of commonly used "A" axis).

As an example:
Gcode loaded :   G1 A66.32041 F2400.00000
Gcode sent to GRBL by gsender: G1 Y66.320 F2400.00000        ("A" has been replaced by "Y")

For this reason, gsender cannot be used for 3D printing.
Is it possible to add an option to handle the A axis as a normal axis (i.e. no change in the gcode) ?
Thanks

walidkayhan, 2025-01-06:
Thank you for the feedback, there were a few other users who had requested this exact feature and we have put it in our backlog of things we want to implement at some point in the future. But for now, it will not be a priority as we are focusing our efforts on other tasks :)

SienciLabs, 2025-02-26:
@fred-c1 just to elaborate in my understanding, if you connect to your board using 'grblHAL' instead of 'grbl', then by default A-axis commands should be supported to be sent without them being modified (just don't turn on rotary mode), would that work for your setup? The only reason for grbl we modify the A-axis commands into Y is because vanilla grbl isn't capable or processing A-axis commands, so we had to create a workaround to allow regular grbl devices to still support "pseudo-rotary-axis" functionality

fred-c1, 2025-03-03:
@SienciLabs    Thanks for looking into this. I tried your suggestion this weekend. Unfortunately, connecting using 'grblHAL' does not work with GRBL mega 5X firmware. After pressing connect, Gsender is frozen during about 10 seconds and then displays "Error 2 - Bad number format".

SienciLabs, 2025-03-03:
@fred-c1 ah I see. I guess GRBL mega 5X is falling between grbl and grblHAL in terms of having grblHAL functionality but grbl-style connection process. In that case it might be tricky for us to come up with a hybrid approach to supporting the mega 5X approach

alexandergrib, 2025-07-04:
@SienciLabs I have a similar issue with my Sainsmart 3030 Prover Max machine.
My GRBL controller is capable of running a true A-axis. But when selecting grblHAL, I'm getting the same error as @fred-c1: "Error 2 - Bad number format". Can you make a toggle switch to select between a true 4th axis for GRBL controllers and vanilla controllers with A conversion to Y axis?

alexandergrib, 2025-07-04:
@hamanjam Not available if connecting with GRBL controller. Only visible if connecting with GRBLHAL controller. Using GRBLHAL I'm getting "Error 2 - Bad number format" error. Also $$ command doesnt bring settings, only OK displayed.

[image]

kglovern, 2026-03-10:
This is available now in 1.6.0-Edge-6 and 1.6.0 (releasing soon) as a configuration option for legacy grbl devices.