Sienci-Labs/gsender #588: Line Number Mismatch in G-Code Error Reporting

siganberg, 2024-12-23:
**Description:**  
When an error occurs, the reported line number of the culprit G-code does not match the actual line number in the G-code file. Even after cleaning up comments and empty lines, the line numbers still do not align.  

This mismatch makes it challenging to troubleshoot and accurately determine where to resume the job.  

**Expected Behavior:**  
The reported line number should match the actual line number in the G-code file for easier troubleshooting and job resumption.  

**Impact:**  
This discrepancy adds difficulty in pinpointing the exact location of issues and determining where to resume operations after an error.  

**Suggestion:**  
Ensure the line numbers reported in errors accurately match the G-code file’s actual line numbers, regardless of comments or empty lines.  

**Additional Context:**  
None.  

Let me know if any further details are required!  

[image]

siganberg, 2024-12-23:
Aside from mismatched line numbers when you have comments or empty lines, it actually reports the wrong g-code line that caused the problem.

As shown in the image below, the actual invalid code is on line 21.

[image]

Here a simple g-code that I use to test 

```
(2D Contour1)
G90 G94
S5000 M3
G54
M8
G0 X-20.316 Y-17.472
Z15
G00 Z5.0 (Changed from: "G1 Z5 F5000")
G01 Z0.632 F3000 (Changed from: "Z0.632 F3000")
Z-19.684
Y-17.48 Z-19.754 F5000
Y-17.503 Z-19.821
Y-17.541 Z-19.881
Y-17.591 Z-19.931
Y-17.651 Z-19.969
Y-17.718 Z-19.992
Y-17.788 Z-20
Y-18.104
G3 X-20 Y-18.42 I0.316 J0

G100 X38.42 ;intended wrong

G1 X38.42
Y18.42
X-38.42
Y-18.42
X-20
G3 X-19.684 Y-18.104 I0 J0.316
G1 Y-17.788
Y-17.718 Z-19.992
Y-17.651 Z-19.969
Y-17.591 Z-19.931
Y-17.541 Z-19.881
Y-17.503 Z-19.821
Y-17.48 Z-19.754
Y-17.472 Z-19.684
G00 Z15.0 (Changed from: "Z15")
```

mattydboom, 2025-01-09:
I have noticed this behaviour too and after some wild goose chases to eventually realise the true culprit was not the code line indicated, but one several lines prior.

The error reporting function is a brilliant and useful idea, but does needs to be made more accurate if it's going to be trusted & fulfil its function properly.

SienciLabs, 2025-02-26:
Good catch @siganberg @mattydboom, I'll add as well, I wonder if this mismatched line reporting has anything to do with @tcurdt 's issue reported here (or could just be a coincidence of two issues both dealing with line counting) https://github.com/Sienci-Labs/gsender/issues/512

terjeio, 2025-04-14:
Best would be to add line numbers to the gcode file...

SienciLabs, 2025-07-22:
Bump this issue to ensure we address it

walidkayhan, 2025-08-08:
Looks like this is an ongoing bug that I have added to our backlog on issues to address, we should hopefully be able to patch it on the next release or two.