Sienci-Labs/gsender #591: Safe Height operation for additional gSender operations

Rbsox18, 2024-12-29:
Setting the Safe Height for gSender works great for front panel button press gotos. I think another useful addition for Safe Height actions would be having Safe Height also apply to any User Macro actions that use a rapid Goto command or any Console user input that uses a rapid Goto command. This would help prevent those broken bit issues when we are focused on the machine operation and not so much on where those clamps are still located.

SienciLabs, 2025-02-26:
@Rbsox18 I think that's a fair idea, but I think what I'd be worried about is opening up a can of worms. The Console and Macros are generally considered to be used for more explicit control of the machine via raw g-code entry, so I'd be worried that if we started to intercept and interfere with those raw commands, that it might cause knock-on issues where someone experiences unexpected movements that they can't stop from happening

Could you give some examples of what situations you're imagining would make sense to add a safe height to the Console or Macros?

Rbsox18, 2025-02-27:
I guess if one is smart enough to enter g-code or write a macro, one should be smart enough make sure his Z axis moves to the top before moving in X or Y…I get that…but the best laid plans and lots of CNC distractions sometimes leads to a small mistake that causes big problems….this was on my mind when I suggested this. Ultimately it is your call.Sent from Robert’s iPad Pro

SienciLabs, 2025-03-01:
Either way we appreciate your input :) I think because Console and Macros are a more raw and unfiltered way of interacting with the CNC, it'll be better to leave them unaffected by any filtering or safety. Maybe in the future though we could recommend safe height movements when you're making Macros by having some sort of smart recognition of the movements you've entered.

Anyway, thanks for your idea and if you have any other then definitely feel free to submit them

closed 2025-03-01 by SienciLabs