Sienci-Labs/gsender #592: Show a startup splashscreen sooner

gcormier, 2025-01-05:
I'm not sure if this is feasible or not.

It would be nice for gSender to show a splash screen as the first/immediate action that takes place.

I find that when clicking on the gSender icon, there's a significant delay before I get any feedback (splash screen), which leads me to wonder if I even clicked on it or not (notably on touch screen).

This happened previously on my HP EliteDesk 800 G2 (i5-6500T), and I've now upgraded to a Beelink EQR6 6600H with the same behavior. Both on Windows 11.  **Even on my main desktop (i7-13700KF), it's over 10 seconds for a splash screen to appear.**

I would imagine the gControl computer might have similar symptoms. It's far worse on a touch screen since you don't have mouse-click feedback and you start the 2nd guess if your touch was registered on the monitor.

walidkayhan, 2025-01-06:
We know this problem has plagued gSender for quite a while. Some old technologies are being used and some throttling caused by Windows Defender's rigorous application checks cause the slow splash screen startup. 

We are looking to improve this in the future and have added it to our backlog, but for the time being, we are focused on other pressing tasks for gSender, hope that makes sense :)

hamanjam, 2025-01-07:
That actually tracks.  I had
McAfee when I first got my miniPC and from power on to gSender opened was 54 seconds and I have since cancelled Mcaffee and have just been using Defender until I figure out what I want to use.  I’ve noticed gSender was taking longer to start and didn’t make the connection.

hamanjam, 2025-01-07:
I just verified I see this issue with Windows Defender.  gSender 1.4.11 takes 43 seconds to show the splash screen.  

If I go into Settings, Windows Security, Virus and Threat Protection, Manage Settings (Under virus and threat protection), Add or remove Exclusions and add the folder c:\Program Files\gSender

gSender now starts in 11 seconds.  Not recommending that anyone exclude anything or disable antivirus, I am just stating that it worked for me.  

[image]

mattydboom, 2025-01-08:
@hamanjam thanks for this great tip, I did the same as now gSender starts within 3-4 seconds as opposed to 10x that previously. 

Also added C:\Program Files\gSender Edge to the list of exclusions whilst there and that too starts just as fast

Cheers :)

SienciLabs, 2025-01-14:
Thanks all for the conversation on this :) I thought we'd previously added this quirk to our documentation but realized I was incorrect so thanks to @hamanjam showing his fix and both him and @mattydboom doubly verifying the fix I've now added it to the docs so that hopefully we can help others too: https://resources.sienci.com/view/gs-installation/#open-faster

Ultimately if we can find a way to not have this issue happen in the first place it'd be ideal, but in the meantime this is probably the best solution

SienciLabs, 2025-07-08:
Hi all just to update, @walidkayhan just did some more work as of today that should speed up the splash screen opening for gSender a little more still, and you should all be able to see this once 1.5.2 is released.

All our other research pointed to the fact that any other improvements would have to come from some major restructuring that we just can't justify right now - but I think between the tip on security management as well as this new tweak, the original issue had now been addressed enough that I'll close it as completed for now. Thanks for everyone pitching in!

closed 2025-07-08 by SienciLabs